nopnop2002/esp-idf-st7789 #10: Blank Display

DiegoPaezA, 2020-04-04:
Hi, I'm trying to use the library but I can't make it work. I already did the configuration as you show in the pictures, the program compile and runs well but never shows anything on the display. I also test the display with and Arduino library and works perfect. Could you give some idea to make it work? Thanks.

nopnop2002, 2020-04-05:
Hello.
As far as I know, ST7789 needs BackLight.

DiegoPaezA, 2020-04-05:
Hi, 

I try it, but nothing change. 
Thanks for your help.

nopnop2002, 2020-04-05:
Hello.

I want to see a picture of your TFT.

DiegoPaezA, 2020-04-05:
Hi, 

I tested the display with tft_espi library (https://github.com/Bodmer/TFT_eSPI) with Arduino using the same connections (without the BLK pin) and works perfect. 

Thanks for your help. 

[image]
[image]

nopnop2002, 2020-04-05:
Same as my TFT.

It is strange why it works without connecting BackLight.

DiegoPaezA, 2020-04-05:
I already did both test, connecting BLk to 3.3v and ground the only change is the display turning on and off. 

[image]

nopnop2002, 2020-04-05:
It's mine.

The CPU frequency is different for some reason.

```
freeRTOS version:V8.2.0
NEWLIB version:3.0.0
lwIP version:2-1-3-0
ESP-IDF version:v4.2-dev-701-g0ae960f2f-dirty
ets_get_cpu_frequency:240
```

DiegoPaezA, 2020-04-05:
I'm going try updating the idf. 

[image]

nopnop2002, 2020-04-05:
The toolchain used from SP-IDF Ver4.1 has been changed to esp32-2019r2.

```
$ xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

DiegoPaezA, 2020-04-05:
After update idf to 4.2, I test the display software and all the examples words as mean to be, but, then i try with a simple blink led and after try again the display software, nothing happen. I really don't understand what is the problem, I clean the project, build all again, and nothing. 
[image]

nopnop2002, 2020-04-05:
Please change the SPI frequency and try.

```
//static const int SPI_Frequency = SPI_MASTER_FREQ_20M;
//static const int SPI_Frequency = SPI_MASTER_FREQ_26M;
static const int SPI_Frequency = SPI_MASTER_FREQ_40M;
//static const int SPI_Frequency = SPI_MASTER_FREQ_80M;
```

DiegoPaezA, 2020-04-06:
Thanks for your Help, finally works.
I change the frequency as you said.  

[image]

closed 2020-04-06 by DiegoPaezA

nopnop2002, 2020-04-06:
I am glad to hear your report.

In the next release, I will change the default CPU frequency to 20Mhz.