nopnop2002/esp-idf-st7789 #14: TFT no powered on for Lilygo TTGO T-Watch-2020

beoran, 2020-10-01:
I am trying to use your library with a Lilygo TTGO T-Watch-2020
https://www.tindie.com/products/ttgo/lilygor-ttgo-t-watch-2020/
I am using thecorrect pins and I tried several SPI frequencies but the display is not powered on, even if I set the backlight

The lvgl esp32 port here:
https://github.com/lvgl/lv_port_esp32/blob/master/components/lvgl_esp32_drivers/lvgl_tft/st7789.c
does work correctly, so I assume something is wrong with the device initialization.
How could I find out what is going wrong?

beoran, 2020-10-01:
It turns out that the backlight IO pin needs to be configured before use with 
gpio_pad_select_gpio(CONFIG_BL_GPIO);
gpio_set_direction(CONFIG_BL_GPIO, GPIO_MODE_OUTPUT);
(both are required)
The lvgl esp32 does this automatically, perhaps it should be added to this library as well? Shall I make a PR?

beoran, 2020-10-01:
I made  #15 for this. Please consider it.

nopnop2002, 2020-10-02:
Thank you for PR.
I have merged your PR.

beoran, 2021-03-02:
Closing this issue sine the PR was merged.

closed 2021-03-02 by beoran