nopnop2002/esp-idf-st7789 #20: Preventing black screen while refreshing text

unre4l, 2021-03-29:
Hi,
i measure values every second an print them onto the screen. To clear old values i first fill the screen black and then draw the text. That results in a visible "refresh" effect, everything is black and the text slowly appears again line per line.  
Is it possible to draw everything in a frame buffer and then paint the screen at once? So that there isn't a completly black screen for one frame. Thanks and cool library!

nopnop2002, 2021-03-29:
>Is it possible to draw everything in a frame buffer and then paint the screen at once?

What is a frame buffer? ??
Do you mean all Pixel data? ??

unre4l, 2021-03-29:
> Do you mean all Pixel data?

Kind of. My goal is to have:
```
Frame 1     ->   Frame 2
| text A |       | text B  |  
```

But what is possible is:
```
(draw text)      (draw clearing rect)       (draw text)
Frame 1     ->   Frame 2               ->   Frame 3
| text A |       |        |                 | text B  |  
```

> If you want to partially rewrite the text on the screen, write the same text in the same color as the background and it will disappear.

Thanks for the example, but unfortunately it wont solve the problem. The frame inbetween the text draw calls has to be a black rectangle (or black text) to clear the previous text. So there is always one frame before new text can appear. 
If one could write all pixel data in a buffer[screen_x][screen_y] and swap all pixel in one draw call, a refresh frame to clear old data wont be necessary, because one can clear and draw on the buffer and flush at once to the screen so no intermediate frame would be visible.

nopnop2002, 2021-03-29:
>If one could write all pixel data in a buffer[screen_x][screen_y] and swap all pixel in one draw call, 

There is lcdDrawMultiPixels()

```
#define PIXELS_Y 240/8
#define PIXELS_X 240

uint8_t PixelData[PIXELS_X][PIXELS_y];

Make a PixelData by yourself.
You can also create the font pattern for the characters yourself.

lcdDrawMultiPixels(TFT_t * dev, 0, 0, PIXELS_X * PIXELS_Y, PixelData);
```

unre4l, 2021-03-30:
The `lcdDrawMultiPixels` solves my problem. Thanks!
I scrolled through your code and found an even simpler solution. 

https://github.com/nopnop2002/esp-idf-st7789/blob/42705cd3a0c99f917f37920db310e5d22303b0e1/main/st7789.c#L772 This line does exactly what i was looking for. Although refreshing is slower, there is no flickering! 👍

nopnop2002, 2021-03-30:
You need to disable this.

https://github.com/nopnop2002/esp-idf-st7789/blob/42705cd3a0c99f917f37920db310e5d22303b0e1/main/st7789.c#L751

nopnop2002, 2021-03-31:
This is done if _font_fill is valid:

https://github.com/nopnop2002/esp-idf-st7789/blob/42705cd3a0c99f917f37920db310e5d22303b0e1/main/st7789.c#L751

closed 2024-12-09 by nopnop2002