nopnop2002/esp-idf-st7789 #25: Can not auto reset on 没有

chenjr15, 2021-12-25:


nopnop2002, 2021-12-28:
The reset timing may be different for each TFT.

nopnop2002, 2022-05-09:
I think there is an electrical problem around reset circuit in the breakout without the CS pin.
Perhaps the reset circuit requires a lot of current.
I inserted a 100 ohm resistor between Vcc and RESET.   

[image]

DaveDavenport, 2022-11-02:
I find the 100 ohm pull up weird.. Its an active low signal and if I measure the level when not asserted it is 3.3V.
 On the esp8266 the display works (without external components, with the adafruit library) and in general esp8266 (from what I found) can sink/source less current then the esp32. I wonder if it is more a timing issue.

Secondly, with the 100 Ohm pull up the esp32 needs to sink at least 33mA to pull the signal low (active low to reset), this feels high. The documentation (but it is not very clear) states a max of 28mA.

If I find time I will hook up a scope and see if I can spot a difference so we know for sure.

DaveDavenport, 2022-11-02:
This made it work for me (tm), might be a double thing?

```patch
diff --git a/main/st7789.c b/main/st7789.c
index 5bde9bc..da8b62e 100644
--- a/main/st7789.c
+++ b/main/st7789.c
@@ -63,11 +63,11 @@ void spi_master_init(TFT_t * dev, int16_t GPIO_MOSI, int16_t GPIO_SCLK, int16_t
                gpio_reset_pin( GPIO_RESET );
                gpio_set_direction( GPIO_RESET, GPIO_MODE_OUTPUT );
                gpio_set_level( GPIO_RESET, 1 );
-               delayMS(50);
+               delayMS(100);
                gpio_set_level( GPIO_RESET, 0 );
-               delayMS(50);
+               delayMS(200);
                gpio_set_level( GPIO_RESET, 1 );
-               delayMS(50);
+               delayMS(300);
        }

        ESP_LOGI(TAG, "GPIO_BL=%d",GPIO_BL);
@@ -98,7 +98,7 @@ void spi_master_init(TFT_t * dev, int16_t GPIO_MOSI, int16_t GPIO_SCLK, int16_t
        memset(&devcfg, 0, sizeof(devcfg));
        devcfg.clock_speed_hz = SPI_Frequency;
        devcfg.queue_size = 7;
-       devcfg.mode = 2;
+       devcfg.mode = 3;
        devcfg.flags = SPI_DEVICE_NO_DUMMY;

        if ( GPIO_CS >= 0 ) {
@@ -199,11 +199,13 @@ bool spi_master_write_colors(TFT_t * dev, uint16_t * colors, uint16_t size)
        return spi_master_write_byte( dev->_SPIHandle, Byte, size*2);
 }
```
change spi mode to 3 (hint I found somewhere else) and add a bit more delay.

nopnop2002, 2022-11-03:
I tried your code, But don’t work for me.

After adding the circuit below, it works fine.

```
ESP32 3V3  -------------------------------+
                                          |
                                          | Emitter
                                         /
                                        /
ESP32 RES  ------------R(2.2K)---------|    S8080
                                Base    \
                                         \
                                          | Collector
                                          |
ST7789 RST -------------------------------+
                                          |
                                          |
                                          |
                                          |
ESP32 GND  ------------R(220)-------------+
```

DaveDavenport, 2022-11-03:
So you are inverting the reset signal? 
(if V(E-B) is above threshold it conducts, so if res is pulled low, ST7780 RST goes high)

nopnop2002, 2022-11-03:
>you are inverting the reset signal?

No.

The two circuits below are very similar, but the current flowing through the RES of the ST7789 is different.
The upper circuit carries more current.

```
ESP32 3V3  -------------------------------+
                                          |
                                          | Emitter
                                         /
                                        /
ESP32 RES  ------------R(2.2K)---------|    S8080
                                Base    \
                                         \
                                          | Collector
                                          |
ST7789 RST -------------------------------+
                                          |
                                          |
                                          |
                                          |
ESP32 GND  ------------R(220)-------------+

ESP32 RES  -------------------------------+
                                          |
ST7789 RST -------------------------------+
```

DaveDavenport, 2022-11-03:
hmm I assume C/E is swapped, so you are pulling the rst to ~(3.3V-0.7V ) .  It is still odd behavior, I will scope it when I have time.

For me the SPI mode 3 (same as was used in the adafruit demo on the esp8266) and slightly longer reset (again picked values from working esp8266 example) works reliable so far. I've done 40 resets and around 15 power cycles and it came back fine every time.

nopnop2002, 2022-11-03:
For me, SPI mode 3 and mode 2 both work fine.

DaveDavenport, 2022-11-03:
I'm going to get some more modules (slighlty different form factor) soon, I will report back how they work.

nopnop2002, 2022-11-03:
If you have a scope, watch the current flowing through RES.

DaveDavenport, 2022-11-03:
[image]
That signal looks fine.  the esp32c3 has no problems asserting that.

nopnop2002, 2022-11-03:
Do you know the current value?

DaveDavenport, 2022-11-03:
Given the above trace, I see no reason to measure current. It clearly is not an issue.

nopnop2002, 2022-11-03:
>Given the above trace, I see no reason to measure current. It clearly is not an issue.

circuit A work fine.
circuit B never work.

can you explain?

circuit A
```

ESP32 3V3  -------------------------------+
                                          |
                                          | Emitter
                                         /
                                        /
ESP32 RES  ------------R(2.2K)---------|    S8050
                                Base    \
                                         \
                                          | Collector
                                          |
ST7789 RST -------------------------------+
                                          |
                                          |
                                          |
                                          |
ESP32 GND  ------------R(220)-------------+
```

circuit B
```
ESP32 RES  -------------------------------+
                                          |
ST7789 RST -------------------------------+
```

nopnop2002, 2022-11-04:
The schematic was wrong.
Sorry for the confusion.

```
ESP32 3V3  -------------------------------+
                                          |
                                          | Collector
                                         /
                                        /
ESP32 RES  ------------R(2.2K)---------|    SS8050
                                Base    \
                                         \
                                          | Emitter
                                          |
ST7789 RST -------------------------------+
                                          |
                                          |
                                          |
                                          |
ESP32 GND  ------------R(220)-------------+
```

DaveDavenport, 2022-11-04:
Ill retry it with an esp32.  Was using an esp32c3 now.

sl45sms, 2023-11-16:
Try to use SPI mode 3 for modules with CS grounded. Not any external circuit required.

DaveDavenport, 2023-11-16:
> Try to use SPI mode 3 for modules with CS grounded. Not any external circuit required.

yes, see my reply above. SPI mode 3 works perfectly reliable.  I have ~400 devices now running with the right mode without any issues or any extra reset circuitry.

nopnop2002, 2023-11-16:
I'll test it on the weekend.

nopnop2002, 2023-11-16:
Changed SPI mode from 2 to 3.
Test results are good.

DaveDavenport, 2023-11-16:
related https://github.com/nopnop2002/esp-idf-st7789/issues/38

nopnop2002, 2023-11-16:
@DaveDavenport 

Thank you.

DaveDavenport, 2023-11-16:
Thanks again for your library.