nopnop2002/esp-idf-st7789 #26: No demo appears on screen

javiser, 2022-01-28:
Hi,
I have been trying to run this demo on an 1.3" display with ST7889 driver connected to a ESP32C3 board. I believe that the pin connections are fine because I managed to get some small example running on Rust with the same wiring. I get this output in the monitor (debug level activated). I don't see any error, but something is still not right. 
```
Executing action: monitor
Serial port /dev/ttyUSB0
Connecting....ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16c8
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d40
entry 0x403ce000
I (30) boot: ESP-IDF v4.4-84-g257d95fc22-dirty 2nd stage bootloader
I (30) boot: compile time 22:34:31
I (30) boot: chip revision: 3
I (33) boot.esp32c3: SPI Speed      : 80MHz
I (38) boot.esp32c3: SPI Mode       : DIO
I (43) boot.esp32c3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot:  3 storage          Unknown data     01 82 00110000 000f0000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0a5d8h ( 42456) map
I (113) esp_image: segment 1: paddr=0001a600 vaddr=3fc8be00 size=0157ch (  5500) load
I (116) esp_image: segment 2: paddr=0001bb84 vaddr=40380000 size=04494h ( 17556) load
I (127) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=284e4h (165092) map
I (158) esp_image: segment 4: paddr=0004850c vaddr=40384494 size=07804h ( 30724) load
I (164) esp_image: segment 5: paddr=0004fd18 vaddr=50000010 size=00010h (    16) load
I (167) boot: Loaded app from partition at offset 0x10000
I (169) boot: Disabling RNG early entropy source...
I (186) cpu_start: Pro cpu up.
D (186) efuse: In EFUSE_BLK2__DATA4_REG is used 3 bits starting with 0 bit
D (187) efuse: In EFUSE_BLK2__DATA4_REG is used 8 bits starting with 12 bit
D (193) efuse: In EFUSE_BLK1__DATA3_REG is used 3 bits starting with 18 bit
D (200) efuse: In EFUSE_BLK1__DATA5_REG is used 5 bits starting with 5 bit
D (207) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 7 bit
D (214) efuse: In EFUSE_BLK1__DATA4_REG is used 7 bits starting with 14 bit
D (221) efuse: In EFUSE_BLK1__DATA4_REG is used 8 bits starting with 21 bit
D (228) efuse: In EFUSE_BLK1__DATA4_REG is used 3 bits starting with 29 bit
D (235) efuse: In EFUSE_BLK1__DATA5_REG is used 5 bits starting with 0 bit
D (249) clk: RTC_SLOW_CLK calibration value: 3775962
I (258) cpu_start: Pro cpu start user code
I (258) cpu_start: cpu freq: 160000000
I (258) cpu_start: Application information:
I (261) cpu_start: Project name:     st7789
I (266) cpu_start: App version:      bd9dcac-dirty
I (271) cpu_start: Compile time:     Jan 28 2022 22:34:30
I (277) cpu_start: ELF file SHA256:  dc6250632bdfbf57...
I (283) cpu_start: ESP-IDF:          v4.4-84-g257d95fc22-dirty
D (290) memory_layout: Checking 4 reserved memory ranges:
D (295) memory_layout: Reserved memory range 0x3fc80000 - 0x3fc8be00
D (302) memory_layout: Reserved memory range 0x3fc8be00 - 0x3fc8ea10
D (308) memory_layout: Reserved memory range 0x3fcdf060 - 0x3fce0000
D (315) memory_layout: Reserved memory range 0x50000000 - 0x50000020
D (321) memory_layout: Building list of available memory regions:
D (327) memory_layout: Available memory region 0x3fc8ea10 - 0x3fca0000
D (334) memory_layout: Available memory region 0x3fca0000 - 0x3fcc0000
D (340) memory_layout: Available memory region 0x3fcc0000 - 0x3fcdf060
D (347) memory_layout: Available memory region 0x50000020 - 0x50002000
I (354) heap_init: Initializing. RAM available for dynamic allocation:
D (361) heap_init: New heap initialised at 0x3fc8ea10
I (366) heap_init: At 3FC8EA10 len 000315F0 (197 KiB): DRAM
I (372) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
D (379) heap_init: New heap initialised at 0x50000020
I (384) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
D (391) FLASH_HAL: extra_dummy: 0
D (394) spi_flash: trying chip: issi
D (398) spi_flash: trying chip: gd
D (401) spi_flash: trying chip: mxic
D (405) spi_flash: trying chip: winbond
D (409) spi_flash: trying chip: boya
D (412) spi_flash: trying chip: generic
I (416) spi_flash: detected chip: generic
I (421) spi_flash: flash io: dio
D (425) cpu_start: calling init function: 0x4200002a
D (430) intr_alloc: Connected src 39 to int 2 (cpu 0)
I (435) sleep: Configure to isolate all GPIO pins in sleep state
I (442) sleep: Enable automatic switching of GPIO sleep configuration
D (449) intr_alloc: Connected src 50 to int 3 (cpu 0)
I (454) cpu_start: Starting scheduler.
D (458) intr_alloc: Connected src 37 to int 4 (cpu 0)
D (458) heap_init: New heap initialised at 0x3fcc0000
D (458) intr_alloc: Connected src 33 to int 7 (cpu 0)
I (468) ST7789: Initializing SPIFFS
D (468) partition: Loading the partition table
D (468) partition: Partition table MD5 verified
I (498) ST7789: Partition size: total: 896321, used: 313499
I (498) SPIFFS_Directory: d_name=esp_logo.png d_ino=0 d_type=1
I (498) SPIFFS_Directory: d_name=LATIN32B.FNT d_ino=0 d_type=1
I (508) SPIFFS_Directory: d_name=ILMH32XB.FNT d_ino=0 d_type=1
I (518) SPIFFS_Directory: d_name=image.bmp d_ino=0 d_type=1
I (528) SPIFFS_Directory: d_name=ILMH16XB.FNT d_ino=0 d_type=1
I (528) SPIFFS_Directory: d_name=esp32.jpeg d_ino=0 d_type=1
I (538) SPIFFS_Directory: d_name=ILGH16XB.FNT d_ino=0 d_type=1
I (538) SPIFFS_Directory: d_name=ILMH24XB.FNT d_ino=0 d_type=1
I (548) SPIFFS_Directory: d_name=ILGH32XB.FNT d_ino=0 d_type=1
I (558) SPIFFS_Directory: d_name=ILGH24XB.FNT d_ino=0 d_type=1
I (568) ST7789: GPIO_CS=-1
I (568) ST7789: GPIO_DC=8
I (568) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (578) ST7789: GPIO_RESET=9
I (578) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
D (588) ST7789: ms=50 _ms=59 portTICK_PERIOD_MS=10 xTicksToDelay=5
D (648) ST7789: ms=100 _ms=109 portTICK_PERIOD_MS=10 xTicksToDelay=10
D (748) ST7789: ms=50 _ms=59 portTICK_PERIOD_MS=10 xTicksToDelay=5
I (798) ST7789: GPIO_BL=18
I (798) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (798) ST7789: GPIO_MOSI=7
I (798) ST7789: GPIO_SCLK=6
D (798) gdma: new group (0) at 0x3fc909b4
D (808) gdma: new pair (0,0) at 0x3fc909e0
D (808) gdma: new tx channel (0,0) at 0x3fc9234c
D (818) gdma: new rx channel (0,0) at 0x3fc90a00
D (818) spi: SPI2 use iomux pins.
D (818) ST7789: spi_bus_initialize=0
D (828) intr_alloc: Connected src 19 to int 9 (cpu 0)
D (828) spi_hal: eff: 20000, limit: 80000k(/0), 0 dummy, -1 delay
D (838) spi_master: SPI2: New device added to CS5, effective clock: 20000kHz
D (848) ST7789: spi_bus_add_device=0
D (848) ST7789: ms=150 _ms=159 portTICK_PERIOD_MS=10 xTicksToDelay=15
D (1008) ST7789: ms=255 _ms=264 portTICK_PERIOD_MS=10 xTicksToDelay=26
D (1268) ST7789: ms=10 _ms=19 portTICK_PERIOD_MS=10 xTicksToDelay=1
D (1278) ST7789: ms=10 _ms=19 portTICK_PERIOD_MS=10 xTicksToDelay=1
D (1288) ST7789: ms=10 _ms=19 portTICK_PERIOD_MS=10 xTicksToDelay=1
D (1298) ST7789: ms=255 _ms=264 portTICK_PERIOD_MS=10 xTicksToDelay=26
D (1558) ST7789: offset(x)=0 offset(y)=0
D (2108) ST7789: offset(x)=0 offset(y)=0
D (2658) ST7789: offset(x)=0 offset(y)=0
I (3208) FillTest: elapsed time[ms]:1650
D (7208) ST7789: offset(x)=0 offset(y)=0
D (7228) ST7789: offset(x)=0 offset(y)=0
D (7248) ST7789: offset(x)=0 offset(y)=0
I (7258) ColorBarTest: elapsed time[ms]:50
D (11268) ST7789: offset(x)=0 offset(y)=0
I (11548) ArrowTest: elapsed time[ms]:280
D (15548) ST7789: offset(x)=0 offset(y)=0
```
I have followed the instructions on the page, and I am also using ESP-IDF 4.4 for this (I checked out today the `release/v4.4` branch)
```
$ idf.py --version
ESP-IDF v4.4-84-g257d95fc22-dirty
```
Any ideas? Thanks for your help!

nopnop2002, 2022-01-28:
>I (798) ST7789: GPIO_SCLK=6

For some reason, there are development boards that cannot use GPIO06, GPIO08, GPIO09, GPIO19 for SPI clock pins.
According to the ESP32C3 specifications, these pins can also be used as SPI clocks.
I used a raw ESP-C3-13 to verify that these pins could be used as SPI clocks.

javiser, 2022-01-28:
Thanks @nopnop2002 for the very quick reply! I had indeed read that text from the `README.md` file, but since I was able to use that pin for the clock signal in the past, I didn't really want to believe that this problem would be relevant for me. I just changed the SCLK pin (0 instead of 6, I left the others as they were) and now it works! Thank you so much!
For the record: I am using the [ESP-C3-32S-Kit ESP32 WiFi+Bluetooth Development Board](https://www.waveshare.com/esp-c3-32s-kit.htm) with ESP32C3 chip from Ai-Thinker, the one with 2 MB external flash (that may be the reason...). Just in case somebody else has the same problem.

closed 2022-01-28 by javiser

nopnop2002, 2022-01-29:
>I just changed the SCLK pin (0 instead of 6, I left the others as they were) and now it works!

I do not know why.
Probably a hardware problem.