nopnop2002/esp-idf-st7789 #27: This Project can not work on esp32C3

Supowang1989, 2022-02-16:
MOSI ： GPIO10
SCLK ：GPIO7 
CS：-1
DC：GPIO1
RESET：GPIO3
BL： GPIO0 

Can anyone help me check it，Thanks！

nopnop2002, 2022-02-16:
For some reason, there are development boards that cannot use GPIO06, GPIO08, GPIO09, GPIO19 for SPI clock pins.
According to the ESP32C3 specifications, these pins can also be used as SPI clocks.
I used a raw ESP-C3-13 to verify that these pins could be used as SPI clocks.

Supowang1989, 2022-02-16:
Thanks for your reply，CS has connect to GND  on my  board

Supowang1989, 2022-02-16:
> Why CS:-1？
> 
> You need to use CS gpio.
> 
> If you don't use CS gpio, you should always set CS of TFT to LOW.

Thanks for your reply，CS GPIO has connect to GND on my board

nopnop2002, 2022-02-16:
For some reason, there are development boards that cannot use GPIO06, GPIO08, GPIO09, GPIO19 for SPI clock pins.
According to the ESP32C3 specifications, these pins can also be used as SPI clocks.
I used a raw ESP-C3-13 to verify that these pins could be used as SPI clocks.

Try to change SPI Clock pin.

Supowang1989, 2022-02-16:
> > Why CS:-1？
> > You need to use CS gpio.
> > If you don't use CS gpio, you should always set CS of TFT to LOW.
> 
> Thanks for your reply，CS GPIO has connect to GND on my board
[Schematic_ESP32-C3-13-EVB-V1_2022-02-16.pdf](https://github.com/nopnop2002/esp-idf-st7789/files/8076461/Schematic_ESP32-C3-13-EVB-V1_2022-02-16.pdf)

this is my sch file

nopnop2002, 2022-02-16:
>this is my sch file

I don't know reason even if I look at it.

Supowang1989, 2022-02-16:
> > this is my sch file
> 
> I don't know reason even if I look at it.

OK ，still very grateful

closed 2022-02-16 by Supowang1989