nopnop2002/esp-idf-st7789 #32: How to display 1 bit depth BMP image?

nguyenxuantan27112000, 2022-07-26:
Hello, I am using BMPTest function in esp_idf library st7789 but this function only supports displaying BMP image with 24 bit depth. 
Here is the command line:
if((result->dib.depth == 24) && (result->dib.compress_type == 0)){
....................................................
}
How to print 1 bit depth BMP image? Please help me fix it, I'm looking forward to your help.
Thanks.

nopnop2002, 2022-07-26:
Here is a description of the BMP format.

You need to understand the BMP format correctly.

http://wiki.gis.com/wiki/index.php/BMP_file_format

nguyenxuantan27112000, 2022-07-27:
Thanks for your reply, after I researched based on your link I found an example of a 2×2 pixel, 24-bit bitmap (Windows DIB header BITMAPINFOHEADER) with pixel format RGB24 and an example of a 4 ×2 pixels, 32-bit bitmap. I can't find 1-bit bitmap example, can I use bmp header, dib header in your library for my purpose? Currently I want to display QR image on TTGO, but I can only create QR bmp file with 1 bit depth and I need your help. I'm just a student, so I don't have the ability, experience to fully understand, hope you can take the time to help me to complete this project. I will try to learn all your guidance,
Thank you.

nguyenxuantan27112000, 2022-07-27:
I used your library and here are the results. This is a 24 bit depth bmp image
[image]
This is a 1 bit depth bmp image
[image]

nopnop2002, 2022-07-27:
>I used your library and here are the results. This is a 24 bit depth bmp image

I want your 24 bit depth bmp image file.

nguyenxuantan27112000, 2022-07-27:
[qrcode_24bit.zip](https://github.com/nopnop2002/esp-idf-st7789/files/9196702/qrcode_24bit.zip)
The bmp file is inside

nguyenxuantan27112000, 2022-07-27:
I used the paint tool on my PC to adjust the depth of the image from 1 to 24, then I added the font folder to run the program and display it. I have now created a QR code bmp with a depth of 1 bit and saved it to my device's spiffs memory

nopnop2002, 2022-07-27:
When I have time, I add a display of monochrome images (QR code).

nguyenxuantan27112000, 2022-07-27:
> When I have time, I will add a display of monochrome images (QR code).

Thank you for your help, can you tell me how long it will take? Because I also have to finish before the deadline. Sorry if this took your time. If you need anything from me please contact me immediately, I am happy to receive it.

nopnop2002, 2022-07-27:
Maybe next weekend.

nguyenxuantan27112000, 2022-07-27:
> Maybe next weekend.
> 
> I have a job at weekday.

Thank you so much !!!

nguyenxuantan27112000, 2022-07-28:
This is 1 bit depth bmp file. I think you will need it.
[qrcode_1bit.zip](https://github.com/nopnop2002/esp-idf-st7789/files/9204753/qrcode_1bit.zip)

nopnop2002, 2022-07-30:
I added QRTest function()

Try latest.

[image]

nguyenxuantan27112000, 2022-07-30:
Thank you very much, I will try it next monday and give you feedback. Hope they help me.

nguyenxuantan27112000, 2022-08-01:
> I added QRTest function()
> 
> Try latest.
> 
> [image]

Hello, 
I tried running with your bmp file with width and height of 240 pixels and it was successful on TTGO T-Display ESP32. Then I tried to use the image with width and height less than 240 pixels, TTGO T-Display ESP32 got error Guru Meditation Error: Core 1 panic'ed (LoadProhibited).
After I debug I found the command line
if ( (sdbuffer[buffidx] & mask) != 0) colors[index] = WHITE;
I don't know if the memory allocation for the sdbffer pointer is related to the error when I change another bmp file with a size other than 240 pixel
[image]
s.

nopnop2002, 2022-08-01:
I tested with a 120x120 BMP file.
It worked fine.

```
$ identify qrcode.bmp
qrcode.bmp BMP 120x120 120x120+0+0 1-bit sRGB 2c 2050B 0.000u 0:00.000
```

[image]

There is another cause.

nopnop2002, 2022-08-01:
>This is 1 bit depth bmp file. I think you will need it.
[qrcode_1bit.zip](https://github.com/nopnop2002/esp-idf-st7789/files/9204753/qrcode_1bit.zip)

This is currupt file.

```
$ identify qrcode_1bit.bmp
qrcode_1bit.bmp BMP3 105x105 105x105+0+0 1-bit sRGB 2c 1742B 0.000u 0:00.000
identify-im6.q16: length and filesize do not match `qrcode_1bit.bmp' @ error/bmp.c/ReadBMPImage/835.
```

nopnop2002, 2022-08-01:
For some unknown reason, this program does not create the correct BMP.

```
$ identify qrcode_test.bmp
qrcode_test.bmp BMP3 84x84 84x84+0+0 1-bit sRGB 2c 1070B 0.000u 0:00.000
identify-im6.q16: length and filesize do not match `qrcode_test.bmp' @ error/bmp.c/ReadBMPImage/835.
```

nopnop2002, 2022-08-01:
You can fix it using convert,

```
$ convert qrcode_test.bmp qrcode_test2.bmp
convert-im6.q16: length and filesize do not match `qrcode_test.bmp' @ error/bmp.c/ReadBMPImage/835.
$ identify qrcode_test2.bmp
qrcode_test2.bmp BMP 84x84 84x84+0+0 1-bit sRGB 2c 1154B 0.000u 0:00.000
```

nguyenxuantan27112000, 2022-08-01:
Thank you, I will try it !!!

> You can fix it using convert,
> 
> ```
> $ convert qrcode_test.bmp qrcode_test2.bmp
> convert-im6.q16: length and filesize do not match `qrcode_test.bmp' @ error/bmp.c/ReadBMPImage/835.
> $ identify qrcode_test2.bmp
> qrcode_test2.bmp BMP 84x84 84x84+0+0 1-bit sRGB 2c 1154B 0.000u 0:00.000
> ```

nguyenxuantan27112000, 2022-08-01:
> You can fix it using convert,
> 
> ```
> $ convert qrcode_test.bmp qrcode_test2.bmp
> convert-im6.q16: length and filesize do not match `qrcode_test.bmp' @ error/bmp.c/ReadBMPImage/835.
> $ identify qrcode_test2.bmp
> qrcode_test2.bmp BMP 84x84 84x84+0+0 1-bit sRGB 2c 1154B 0.000u 0:00.000
> ```

It worked fine. 
Can you edit the code file so that it generates the correct bmp file?

nopnop2002, 2022-08-01:
>Can you edit the code file so that it generates the correct bmp file?

No.

This is not my issue.

nguyenxuantan27112000, 2022-08-01:
Sorry and thank you for your support.
I want the device to be able to generate QR codes on its own and they can be displayed on the screen, but currently the way to generate QR codes is not correct.
Can you guide me how to generate the correct QR code? When I have a problem, can I contact you ?

nopnop2002, 2022-08-01:
I used on-line qr code generator.

Plz google [QR generator]

>When I have a problem, can I contact you ?

If it's related to this repository, it's possible, but otherwise it's not possible.

nopnop2002, 2022-08-01:
I have a one hint.

It helps if you know how to use the convert command.

closed 2022-08-01 by nopnop2002

nguyenxuantan27112000, 2022-08-01:
please speak

nguyenxuantan27112000, 2022-08-01:
Speak, please.

nopnop2002, 2022-08-01:
Plz google [linux convert]

nopnop2002, 2022-08-01:
It's usually like this:

Create your QR code using online generator

Use the identify command to check the number of colors and size of the BMP

Resize using convert command

Convert to monochrome image using convert command

Please do your own research later.

nguyenxuantan27112000, 2022-08-01:
I think the way you instruct is how to generate a QR code when using a command entered from the keyboard. But I need to write a library so that when loaded into the chip it will automatically generate a QR code when conditions are met.
I wonder is it possible to create a function that functions like the convert command? That seems like an impossibility, right?
I tried convert and it worked. Thanks for this helped me.

nopnop2002, 2022-08-01:
>But I need to write a library so that when loaded into the chip it will automatically generate a QR code when conditions are met.

You can create a QR code BMP file using ESP32 + ESP-IDF.

However, it is inappropriate to include that feature in this repository.

This will be a completely different repository.

It is necessary to consider how to enter the text string that is the basis of the QR code.

nopnop2002, 2022-08-01:
The easiest way is to find a REST server that creates a BMP image of the QR code.

Example:

Send Get request to REST Server:
http://hogehoge.com/create-qr-code/?data=HelloWorld!&size=100x100&depth=1

Receive a BMP image that depth=1 from a REST server using GET.

Save the received BMP image in SPIFFS with the BMP extension.

But I can't find such a REST server.

nguyenxuantan27112000, 2022-08-01:
With my current ability the things you suggest are first known. Currently I have generated an ESP32 + ESP-IDF QR code (I sent it above for you to see) and saved it as a bmp file into the device's SPIFFS memory, but the generated file has problems and it doesn't show up on the screen. I will fix the file save function first, if not I will learn your way.
Thank you very much.

nopnop2002, 2022-08-01:
>Currently I have generated an ESP32 + ESP-IDF QR code (I sent it above for you to see) and saved it as a bmp file into the device's SPIFFS memory, 

Good!!

Publish it on github when you're done.

nguyenxuantan27112000, 2022-08-01:
Okkk !!!