nopnop2002/esp-idf-st7789 #33: Not working in ESP-IDF5.0

ameharoo, 2022-11-04:
I just clone this repo, set pins in menuconfig and no see demo
Display backlight changes correctly, but not displaying demo.
Also: in esp-idf5.0 i had to fix types in ESP_LOG calls.
<details>
  <summary>UART log</summary>

  ```
  I (30) boot: ESP-IDF v5.0-beta1-473-geea8629fa1-dirt 2nd stage bootloader
  I (30) boot: compile time 21:22:17
  I (31) boot: chip revision: V100
  I (35) boot_comm: chip revision: 1, min. bootloader chip revision: 0
  I (42) boot.esp32: SPI Speed      : 40MHz
  I (47) boot.esp32: SPI Mode       : DIO
  I (52) boot.esp32: SPI Flash Size : 2MB
  I (56) boot: Enabling RNG early entropy source...
  I (62) boot: Partition Table:
  I (65) boot: ## Label            Usage          Type ST Offset   Length
  I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
  I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
  I (87) boot:  2 factory          factory app      00 00 00010000 00100000
  I (95) boot:  3 storage          Unknown data     01 82 00110000 000f0000
  I (102) boot: End of partition table
  I (107) boot_comm: chip revision: 1, min. application chip revision: 0
  I (114) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0b7d4h ( 47060) map
  I (139) esp_image: segment 1: paddr=0001b7fc vaddr=3ffb0000 size=02a10h ( 10768) load
  I (144) esp_image: segment 2: paddr=0001e214 vaddr=40080000 size=01e04h (  7684) load
  I (148) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=260f0h (155888) map
  I (210) esp_image: segment 4: paddr=00046118 vaddr=40081e04 size=0b68ch ( 46732) load
  I (229) esp_image: segment 5: paddr=000517ac vaddr=50000000 size=00010h (    16) load
  I (237) boot: Loaded app from partition at offset 0x10000
  I (237) boot: Disabling RNG early entropy source...
  I (249) cpu_start: Pro cpu up.
  I (250) cpu_start: Starting app cpu, entry point is 0x40081148
  I (0) cpu_start: App cpu up.
  I (264) cpu_start: Pro cpu start user code
  I (264) cpu_start: cpu freq: 240000000 Hz
  I (264) cpu_start: Application information:
  I (269) cpu_start: Project name:     st7789
  I (274) cpu_start: App version:      1
  I (278) cpu_start: Compile time:     Nov  4 2022 22:59:37
  I (284) cpu_start: ELF file SHA256:  dd5bdf4f472a3adf...
  I (290) cpu_start: ESP-IDF:          v5.0-beta1-473-geea8629fa1-dirt
  I (297) heap_init: Initializing. RAM available for dynamic allocation:
  I (304) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
  I (310) heap_init: At 3FFB3B38 len 0002C4C8 (177 KiB): DRAM
  I (317) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
  I (323) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
  I (329) heap_init: At 4008D490 len 00012B70 (74 KiB): IRAM
  I (337) spi_flash: detected chip: generic
  I (340) spi_flash: flash io: dio
  W (344) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
  I (358) cpu_start: Starting scheduler on PRO CPU.
  I (0) cpu_start: Starting scheduler on APP CPU.
  I (368) ST7789: Initializing SPIFFS
  I (458) ST7789: Partition size: total: 896321, used: 321782
  I (458) SPIFFS_Directory: d_name=esp32.jpeg d_ino=0 d_type=1
  I (458) SPIFFS_Directory: d_name=esp_logo.png d_ino=0 d_type=1
  I (468) SPIFFS_Directory: d_name=ILGH16XB.FNT d_ino=0 d_type=1
  I (468) SPIFFS_Directory: d_name=ILGH24XB.FNT d_ino=0 d_type=1
  I (478) SPIFFS_Directory: d_name=ILGH32XB.FNT d_ino=0 d_type=1
  I (478) SPIFFS_Directory: d_name=ILMH16XB.FNT d_ino=0 d_type=1
  I (488) SPIFFS_Directory: d_name=ILMH24XB.FNT d_ino=0 d_type=1
  I (498) SPIFFS_Directory: d_name=ILMH32XB.FNT d_ino=0 d_type=1
  I (498) SPIFFS_Directory: d_name=image.bmp d_ino=0 d_type=1
  I (518) SPIFFS_Directory: d_name=LATIN32B.FNT d_ino=0 d_type=1
  I (518) SPIFFS_Directory: d_name=qrcode.bmp d_ino=0 d_type=1
  I (538) ST7789: GPIO_CS=-1
  I (538) ST7789: GPIO_DC=27
  I (548) gpio: GPIO[27]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
  I (548) ST7789: GPIO_RESET=33
  I (548) gpio: GPIO[33]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
  I (738) ST7789: GPIO_BL=32
  I (738) gpio: GPIO[32]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
  I (738) ST7789: GPIO_MOSI=23
  I (738) ST7789: GPIO_SCLK=18
  I (3138) FillTest: elapsed time[ms]:1650
  I (7188) ColorBarTest: elapsed time[ms]:50
  I (11458) ArrowTest: elapsed time[ms]:270
  I (17588) LineTest: elapsed time[ms]:2130
  I (23478) CircleTest: elapsed time[ms]:1890
  I (29408) RoundRectTest: elapsed time[ms]:1930
  I (39408) RectAngleTest: elapsed time[ms]:6000
  I (50398) TriangleTest: elapsed time[ms]:6980
  I (54828) DirectionTest: elapsed time[ms]:430
  I (59878) HorizontalTest: elapsed time[ms]:1050
  I (64928) VerticalTest: elapsed time[ms]:1050
  I (69118) FillRectTest: elapsed time[ms]:190
  I (73358) ColorTest: elapsed time[ms]:240
  I (78378) CodeTest: elapsed time[ms]:1020
  I (84018) CodeTest: elapsed time[ms]:1640
  I (95288) BMPTest: elapsed time[ms]:7270
  I (101858) JPEGTest: elapsed time[ms]:2570
  I (108658) PNGTest: elapsed time[ms]:2800
  I (112848) QRTest: elapsed time[ms]:190
  I (123308) FillTest: elapsed time[ms]:1650
  I (127358) ColorBarTest: elapsed time[ms]:50
  I (131628) ArrowTest: elapsed time[ms]:270
  I (137758) LineTest: elapsed time[ms]:2130
  I (143648) CircleTest: elapsed time[ms]:1890
  I (149578) RoundRectTest: elapsed time[ms]:1930
  I (159578) RectAngleTest: elapsed time[ms]:6000
  I (170568) TriangleTest: elapsed time[ms]:6980
  I (174998) DirectionTest: elapsed time[ms]:430
  I (180048) HorizontalTest: elapsed time[ms]:1050
  I (185098) VerticalTest: elapsed time[ms]:1050
  I (189318) FillRectTest: elapsed time[ms]:220
  I (193558) ColorTest: elapsed time[ms]:240
  I (198578) CodeTest: elapsed time[ms]:1020
  I (204208) CodeTest: elapsed time[ms]:1630
  I (215448) BMPTest: elapsed time[ms]:7240
  ```
</details>

nopnop2002, 2022-11-05:
For the unsigned int32 logging format, an interim workaround is provided here.
https://github.com/espressif/esp-idf/issues/9511
This issue look like a bug in ESP-IDF Ver5 Beta.
It may be fixed in ESP-IDF Ver5 RC1.

The following commit does not cause any problems.
$ git checkout 9b8c558e63d95b491355a22e69f247834e899b47

---

There are two types of ST7789 breakouts.   
One has a CS pin and one does not have a CS pin.   
I think there is an electrical problem __around reset circuit__ in the breakout without the CS pin.   
Perhaps the reset circuit requires a lot of current.   
Ultimately I ended up using the following circuit.   
And press the reset button several times.

```
ESP32 3V3  -------------------------------+
                                          |
                                          | Collector
                                         /
                                        /
ESP32 RES  ------------R(2.2K)---------|    SS8050
                                Base    \
                                         \
                                          | Emitter
                                          |
ST7789 RST -------------------------------+
                                          |
                                          |
                                          |
                                          |
ESP32 GND  ------------R(220)-------------+
```

ameharoo, 2022-11-05:
Thanks for the answer! I tried this connection (through a transistor) and nothing happened.
My circuit is using no-CS connection.
It turns out that this screen is faulty?
PS. I used 2N3904 instead SS8050

nopnop2002, 2022-11-05:
Any NPN transistor will probably do.
press the reset button several times.

mandolins99, 2023-02-25:
Changing  devcfg.mode = 0; works for me.

nopnop2002, 2023-11-16:
Changed SPI mode from 2 to 3.

closed 2023-11-16 by nopnop2002