nopnop2002/esp-idf-st7789 #34: Slow Refresh Rate of ST7789 Screen Compared to ESP-IDF Screen Library

nikhil-robinson, 2023-05-04:
I've been using the ESP-IDF-ST7789 library for my project, and while I love the rendering capabilities of the GFX, I have noticed that the refresh rate of the ST7789 screen is significantly slower compared to the screen library of the ESP-IDF.

I believe that this issue could potentially hinder the performance of certain projects that rely heavily on fast refresh rates. It would be great if there could be some improvements made to the library to enhance the refresh rate of the ST7789 screen and bring it closer to the performance of the ESP-IDF screen library.

Thank you for your hard work on this library, and I look forward to seeing how this issue can be addressed.

nikhil-robinson, 2023-05-04:
screen library : https://github.com/espressif/esp-iot-solution/tree/master/examples/screen

ESP-IDF LCD driver : https://github.com/espressif/esp-idf/tree/56123c52aa/examples/peripherals/lcd/tjpgd

both f theses are very fast regarding decoding and displaying frames for st7789. but what they are lacking is the drawing capabilities which this project has

nopnop2002, 2023-05-04:
thank you

we'll look into these.

nopnop2002, 2023-05-04:
i read this:
screen library : https://github.com/espressif/esp-iot-solution/tree/master/examples/screen

Does this app support ST7789?

nopnop2002, 2023-05-04:
I understand the difference between ESP-IDF LCD driver and mine.

ESP-IDF LCD driver : https://github.com/espressif/esp-idf/tree/56123c52aa/examples/peripherals/lcd/tjpgd

The ESP-IDF LCD driver always draws full pixel.

Redraw the full pixel even if you only update 1 pixel.

This is fast for displaying images, but slow for displaying text and graphics.

closed 2023-06-23 by nopnop2002