nopnop2002/esp-idf-st7789 #40: Variable 'size' overflows in function lcdDrawFinish() when setting the resolution to 320*240

MNDJ666, 2023-12-20:
My 320*240 TFT  can only displays a limited part when enabling frame buffer, then I found that variable 'size' overflowed, where it should be set to uint32_t.  It doesn't work under resolution at 320*240,
 https://github.com/nopnop2002/esp-idf-st7789/blob/8adf642cc26cf5d5815fb09d3006d063cb7c826e/components/st7789/st7789.c#L1069

nopnop2002, 2023-12-20:
Will it work correctly if I change it to uint32_t?

MNDJ666, 2023-12-20:
> Will it work correctly if you change to uint32_t in 320x240 TFT?

Yes, it works perfectly now.
before:
[image]
after:
[image]

nopnop2002, 2023-12-20:
Thank you for the report.
Fixed.

closed 2023-12-20 by nopnop2002