nopnop2002/esp-idf-st7789 #42: Esp32S3: display remains black with backlight only

davidealessandrini88, 2024-06-03:
I've tried using code with ESP32-S3_WROOM1 kit but without success.
GPIO23 is't present in esp32s3 so I changed it.
I've tried different configurations but without results.
No error on idf monitor but display remains black with backlight only.

I've used different displays, the same shown in the README guide.

To reproduce the problem just compile for ESP32-S3 and connect display in some way, I can't find a valid pins combination.

Thank you

nopnop2002, 2024-06-03:
Please utilize this code to check if GPIO is working properly.

https://github.com/espressif/esp-idf/tree/master/examples/get-started/blink

davidealessandrini88, 2024-06-04:
Hello,
I apologize, but after so many tests I had lost sight of the initial configuration. So I tried again with the main configuration. I hooked up a Saleae to see the MOSI and CLK signals and they look good. With the Saleae connected, for once the test worked, I finally saw the display working. However it only happened once.

I tried to shorten the cables from board to display as much as possible but it didn't change, I couldn't get the display to work anymore.

I am now able to confirm that the software works, the problem is probably in the wiring.

I encourage you to provide advice/details, if you have any, on this topic.

Thank you

davidealessandrini88, 2024-06-04:
[image]

nopnop2002, 2024-06-04:
Try pulling up the RES pin.

davidealessandrini88, 2024-06-06:
I tried the pull-up but unfortunately it doesn't solve the problem. At work I will be able to use a good oscilloscope to better analyze the signals.
In any case the software has no problems, this issue can be closed. Thanks for the advice, I hope I can get the device to work.
Best regards
David

nopnop2002, 2024-06-06:
From experience so far, the device seems to require a relatively large current at startup.

It may be effective to convert 5V to 3V3 with a regulator and supply it.

Abluee, 2024-07-07:
> > GPIO23 is't present in esp32s3 so I changed it.esp23s3 中不存在 GPIO32，所以我更改了它。
> 
> you are doing something wrong.你做错了什么。 This is the default for ESP32S3.这是 ESP32S3 的默认设置。 There is No GPIO23.没有 GPIO23。 ![st7789_default](https://private-user-images.githubusercontent.com/6020549/336250437-e8a93e97-e13c-4c7a-a214-50a8ad1af310.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjAzMjYwOTEsIm5iZiI6MTcyMDMyNTc5MSwicGF0aCI6Ii82MDIwNTQ5LzMzNjI1MDQzNy1lOGE5M2U5Ny1lMTNjLTRjN2EtYTIxNC01MGE4YWQxYWYzMTAuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcwNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MDdUMDQxNjMxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDllODFiNDI4OTEyMWIyNzA3YzdhOGU1MmJlNzVhMzc3MGY0OGUwMmEzNjkzNmY1NTBhOTQ3NTllMTZlMDIxOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.aSaTC10PZsUPXU_ZLoSIicmo8BQUYehVkggiQw_IBO0)

I encountered the same issue. However, instead of using the program configuration interface to modify the GPIO pins, I changed the #if 0 to #if 1 in the main.c code for the GPIO configuration used for debugging. After flashing, the program displayed normally. I suspect the issue might be due to the program not reading your GPIO configuration correctly. I hope this helps.

[image]

nopnop2002, 2024-07-07:
@Abluee 

Once it works, I think it will work correctly even if you change it back.

arunyava, 2024-07-31:
[image]
Use these pin configuration in esp32s3  and use external power also .

nopnop2002, 2024-07-31:
Can you share your code to identify the problem?

Pr47h4m, 2024-08-18:
> ![IMG_2601](https://private-user-images.githubusercontent.com/19437918/336576386-029ad285-38e6-474a-8476-e47d0490ed6f.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM5OTc5NDksIm5iZiI6MTcyMzk5NzY0OSwicGF0aCI6Ii8xOTQzNzkxOC8zMzY1NzYzODYtMDI5YWQyODUtMzhlNi00NzRhLTg0NzYtZTQ3ZDA0OTBlZDZmLkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODE4VDE2MTQwOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVmZDcwMzJiYTJjYzI3MzgzZmNjMjc5YjRjYjBiYzM0MWI0Zjg0ZTQ5ZjIxYmNmYTJhYTUxOTAyNjg1NWYxOTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Nu-qbj0lvpVE6qIhfSbcP_3jJuK_2YchhXV7mIZ4dWs)

Hello
I am using ESP32-S3 and same display
I am also encountering same issue as you 
Have you found any solution?
Thanks

closed 2026-08-29 by nopnop2002