nopnop2002/esp-idf-st7789 #43: About create font

nguyentrungg, 2024-06-14:
Hi. Thank you for creating a perfect library.
I want to ask you how to create file .FNT from .ttf file example: concielian.ttf.
Thank you so much.

nopnop2002, 2024-06-14:
TTF fonts are scalable fonts.

FNT fonts are bitmap fonts.

Searching for this keyword brings up several hits, but I have never tried it.

convert scalable fonts to bitmap font

nopnop2002, 2024-06-14:
You can use otf2bdf to convert TTF font files to BDF font files.   
You can use fontedit.exe to convert BDF font files to FONTX format.
```
otf2bdf AGENCYB.TTF > AGENCYB.bdf
```

closed 2024-07-30 by nopnop2002

nguyentrungg, 2024-10-18:
Hi, I saw you post an answer with instructions on how to convert ttf file to fnt file using some tool on ubuntu. I remember it was sudo otf2bdf, but I don't remember clearly. Please remind me, I would appreciate it.

nopnop2002, 2024-10-18:
https://formulae.brew.sh/formula/otf2bdf