nopnop2002/esp-idf-st7789 #44: Custom Font creation

arunyava, 2024-07-24:
Hi,
please help to create custom font .I look fontfx.exe .but it's not found .Can you please explain the steps. I am new in this field

nopnop2002, 2024-07-24:
[FONTX Editor](http://elm-chan.org/fsw/fontxedit.zip)

arunyava, 2024-07-25:
while clicking the link it shows "not found".

arunyava, 2024-07-25:
its downloaded sir. thanks a lot for your valuable support

nopnop2002, 2024-07-25:
>how can we select ttf file for custom font creation.

It's written in the readme.

arunyava, 2024-07-25:
hi sir,
How can make the letter bold

nopnop2002, 2024-07-25:
Please check bold.

arunyava, 2024-07-26:
[image]
this is the exe i got from that link. It has no option. what do you do sir .

arunyava, 2024-07-26:
Hi Sir,
I am facing one more issue. if i set color as yellow but the font letters are in blue.  
                         GetFontx(0, buffer, &fontWidth, &fontHeight);
			  lcdFillScreen(&dev, BLACK);
			  lcdSetFontDirection(&dev, 3);
				color = YELLOW;
				strcpy((char *)ascii, "123456789101234567");
				lcdDrawString(&dev, (fontHeight*1)-1,CONFIG_HEIGHT-12, ascii, color);
what was the issue ? Can you please help me?

nopnop2002, 2024-07-27:
Is BLUE displayed correctly?
```
color = BLUE;
strcpy((char )ascii, "123456789101234567");
lcdDrawString(&dev, (fontHeight1)-1,CONFIG_HEIGHT-12, ascii, color);
```

arunyava, 2024-07-29:
no , it changed as yellow

nopnop2002, 2024-07-29:
```
#define RED    rgb565(255,   0,   0) // 0xf800
#define GREEN  rgb565(  0, 255,   0) // 0x07e0
#define BLUE   rgb565(  0,   0, 255) // 0x001f
#define BLACK  rgb565(  0,   0,   0) // 0x0000
#define WHITE  rgb565(255, 255, 255) // 0xffff
#define GRAY   rgb565(128, 128, 128) // 0x8410
#define YELLOW rgb565(255, 255,   0) // 0xFFE0
#define CYAN   rgb565(  0, 156, 209) // 0x04FA
#define PURPLE rgb565(128,   0, 128) // 0x8010
```

arunyava, 2024-07-30:
Sorry sir , its not your project issue. Its my project issue . I need help to rectify the issue. Thanks in advance. I'm working  in LPC1768 board.

nopnop2002, 2024-07-30:
I don't have LPC1768 board.

nopnop2002, 2024-07-30:
The issues of creating a custom font has been resolved, so I close it.

closed 2024-07-30 by nopnop2002