nopnop2002/esp-idf-st7789 #47: Need to erase some data values

arunyava, 2024-11-07:
Hi,
This post was not and issue.
I need some help for clearing the data. I will write some string value to display. for example i wrote WELCOME' string on display. I want to erase E and M on the display. How can i achieve this. Thanks in advance!

arunyava, 2024-11-20:
Thank you sir for reply. One more help. How to draw a symbol means bluetooth symbol , battery symbols like that? Is it possible to help me to do that?

nopnop2002, 2024-11-20:
A font file is required that contains symbols such as the Bluetooth symbol and battery symbol.

I believe that such font files are available on the Internet.

[image]

arunyava, 2024-11-25:
How can you create this image. Can you simply explain the step.

arunyava, 2024-11-25:
Unable to download FONTX editor

nopnop2002, 2024-11-25:
Perhaps your browser is blocking the download.

nopnop2002, 2024-11-26:
I added IconTest example.
[image]

I created a PNG file with this.
https://github.com/nopnop2002/esp-idf-st7789/tree/master/battery_image

arunyava, 2024-11-29:
Can you explain "This is a 16x16 font converted from u8g2 library Scroll-o-Sprites.bdf." this one?

nopnop2002, 2024-11-29:
Import Scroll-o-Sprites.bdf included in the u8g2 library into fontx editor and save it as an FNT file.

nopnop2002, 2024-12-05:
If you want to change the BDF font size, you need to change the bdf file.

There is probably a tool somewhere to change the font size of bdf files.

arunyava, 2024-12-09:
[font_data.zip](https://github.com/user-attachments/files/18058984/font_data.zip)
I am attaching font file which i am created.

nopnop2002, 2024-12-09:
Too little information.
Information such as screenshots and source code is required.
If you cannot meet my requirements, I will close this issue.

nopnop2002, 2024-12-09:
Thanks for sharing the font file.
Supports large fonts.
[image]

arunyava, 2024-12-10:
Is it possible to share the changes you made  in the latest version. It will very helpful for me to integrate my existing code, because  i added so many other functions.

nopnop2002, 2024-12-10:
>Is it possible to share the changes you made in the latest version. 

https://github.com/nopnop2002/esp-idf-st7789/commit/6a57561c58db739cae676ef0cc6fc2f70bfb8b2b

closed 2024-12-18 by nopnop2002

arunyava, 2024-12-19:
i need help to rotate a PNG image. Currently it placed in one direction .If change the direction of text it will change .But in case of image it didn't changed. Please help me to resolve. Thanks in advance

nopnop2002, 2024-12-19:
Rotating an image on ESP32 takes a very long time.
First load an image that has been rotated using imagemagick.

```
convert before.jpg -rotate +45 after.jpg
```
[image]