nopnop2002/esp-idf-st7789 #51: text can't be displayed

networkjarwy1, 2025-01-27:
I have spent a lot of time reading the main code to understand the library, because it has no documentation. I have tried a lot of things to init the font, etc. and I cant display text, but I can change pixel color. It is weird that test code you provided works perfectly, but my code doesnt work. How is that possible?

nopnop2002, 2025-01-27:
>How is that possible?

Only you can fix the problems in your code.

networkjarwy1, 2025-01-27:
> A good way is to modify the example code I provide bit by bit.
> 

Yeah, I did that. But it doesn't work. May I ask you for a small example of code to display text? I am lost in it and its only thing I need

nopnop2002, 2025-01-27:
>May I ask you for a small example of code to display text?

https://github.com/nopnop2002/esp-idf-st7789/blob/master/main/main.c#L107

closed 2025-02-11 by nopnop2002