nopnop2002/esp-idf-st7789 #52: Need support to display animations and textanimations

arunyava, 2025-01-28:
Hi sir,
I need to your help to create animation and animated text effect. Can you please help me to solve this problem. Thanks in advance.

nopnop2002, 2025-01-28:
Enable FrameBuffer in Menuconfig to run the animation demo.

arunyava, 2025-01-28:
How to move text on the display into different position

nopnop2002, 2025-01-28:
Take the string you want from the FrameBuffer and move it to another position.

arunyava, 2025-01-28:
any method without using framebuffer

nopnop2002, 2025-01-28:
There isn't.

arunyava, 2025-01-31:
Is it possible to use TFT_eSPI library in esp idf for ST7789 LCD

arunyava, 2025-01-31:
Hi can you simply demonstrate ay animation effect in your source code.It will helpful for me

nopnop2002, 2025-01-31:
>Is it possible to use TFT_eSPI library in esp idf for ST7789 LCD

I don't know about other libraries.
Ask the author of the TFT_eSPI library.

nopnop2002, 2025-01-31:
>Hi can you simply demonstrate ay animation effect in your source code.It will helpful for me

This code will help you.

https://github.com/nopnop2002/esp-idf-st7789/blob/master/main/main.c#L1213

arunyava, 2025-01-31:
can you please demonstrate a simple animation for me.It will very helpful for me.

nopnop2002, 2025-01-31:
Simple animations have already been achieved.

https://github.com/nopnop2002/esp-idf-st7789/blob/master/main/main.c#L1213

arunyava, 2025-02-10:
Hi Sir,
i want to display emotion currently it doesn't display . I can't know why it is display?
FontxFile fx32S[2];
    		InitFontx(fx32S,"/fonts/Scroll-o-Sprites.fnt",""); // 16x16Dot Emoji
    		CodeTest(&dev, fx32S, CONFIG_WIDTH, CONFIG_HEIGHT, 0x00, 0xFF);  i am using this piece of code.

nopnop2002, 2025-02-10:
Have you initialized the dev variables?

arunyava, 2025-02-11:
But in your code i ant find Scroll-o-Sprites.fnt

nopnop2002, 2025-02-11:
Are the font files in the fonts directory?