nopnop2002/esp-idf-st7789 #53: Is there a lvgl demo for this library?

NewNoob2002, 2025-03-17:
Hi dear, I want to request a demo using this library, I love this library, I tried to port lvgl last week
But I failed, I used lvgl demo to test, when I use Darwpixel function, it works, but the drawing speed is too low.
So I tried to use drawbuffers function but failed, it shows all white, my screen is 172x320
If you can help me with anything, I will be grateful

nopnop2002, 2025-03-17:
When FrameBuffer is enabled, all output will be stored in the internal FrameBuffer and reflected to the device with lcdDrawFinish.

NewNoob2002, 2025-03-17:
```
static void flush_cb(lv_display_t * disp, const lv_area_t * area, uint8_t * px_map)
{
    if (disp == nullptr || area == nullptr || px_map == nullptr) {
        ESP_LOGE("Lcd", "Invalid pointer in flush_cb");
        return;
    }

    uint32_t w = (area->x2 - area->x1 + 1);
    uint32_t h = (area->y2 - area->y1 + 1);

    uint16_t * buf16 = (uint16_t *)px_map; /*Let's say it's a 16 bit (RGB565) display*/
    ESP_LOGI("lvgl", "w=%ld h=%ld", w, h);
	uint32_t size = w*h;
    lcdDrawMultiPixels(area->x1, area->y1, size, buf16);

    
    lv_display_flush_ready(disp);
}
```
I used this lcdDrawMultiPixels api, you mean use like below?
```
static void flush_cb(lv_display_t * disp, const lv_area_t * area, uint8_t * px_map)
{
    if (disp == nullptr || area == nullptr || px_map == nullptr) {
        ESP_LOGE("Lcd", "Invalid pointer in flush_cb");
        return;
    }

    for(int i = area->x1; i < area->x2; i++)
    {
            for(int j = area->y1; j < area->y2; j++)
            {
                uint16_t color = px_map[(j-area->y1)*area->x2+i-area->x1];
                lcdDrawPixel(i, j, color);
            }
    }
    lcdDrawFinish();
    
   
``` lv_display_flush_ready(disp);
}

nopnop2002, 2025-03-17:
> you mean use like below?

yes.
but need argument.
https://github.com/nopnop2002/esp-idf-st7789/blob/master/components/st7789/st7789.c#L1163

NewNoob2002, 2025-03-18:
> ```
> static void flush_cb(lv_display_t * disp, const lv_area_t * area, uint8_t * px_map)
> {
>     if (disp == nullptr || area == nullptr || px_map == nullptr) {
>         ESP_LOGE("Lcd", "Invalid pointer in flush_cb");
>         return;
>     }
> 
>     uint32_t w = (area->x2 - area->x1 + 1);
>     uint32_t h = (area->y2 - area->y1 + 1);
> 
>     uint16_t * buf16 = (uint16_t *)px_map; /*Let's say it's a 16 bit (RGB565) display*/
>     ESP_LOGI("lvgl", "w=%ld h=%ld", w, h);
> 	uint32_t size = w*h;
>     lcdDrawMultiPixels(area->x1, area->y1, size, buf16);
> 
>     
>     lv_display_flush_ready(disp);
> }
> ```
> 
> I used this lcdDrawMultiPixels api, you mean use like below?我使用了这个 lcdDrawMultiPixels api，您的意思是使用如下？
> 
> ```
> static void flush_cb(lv_display_t * disp, const lv_area_t * area, uint8_t * px_map)
> {
>     if (disp == nullptr || area == nullptr || px_map == nullptr) {
>         ESP_LOGE("Lcd", "Invalid pointer in flush_cb");
>         return;
>     }
> 
>     for(int i = area->x1; i < area->x2; i++)
>     {
>             for(int j = area->y1; j < area->y2; j++)
>             {
>                 uint16_t color = px_map[(j-area->y1)*area->x2+i-area->x1];
>                 lcdDrawPixel(i, j, color);
>             }
>     }
>     lcdDrawFinish();
>     
>    
> ``` lv_display_flush_ready(disp);
> }
> ```

NewNoob2002, 2025-03-18:
i'm sorry, Accidentally pressed the close button

nopnop2002, 2025-03-18:
I don't understand the meaning of your question.

NewNoob2002, 2025-03-18:
ok, Maybe I didn't explain it clearly. What I meant is that if the frame buffer is used, it's same in the two situations below?
```
    for(int i = area->x1; i < area->x2; i++)
    {
            for(int j = area->y1; j < area->y2; j++)
            {
                uint16_t color = px_map[(j-area->y1)*area->x2+i-area->x1];
                lcdDrawPixel(dev, i, j, color);
            }
    }
    lcdDrawFinish();
```
```
lcdDrawFillRect(dev, area->x1, area->y1, area->x2-1, area->y2-1, color);
lcdDrawFinish();
```

nopnop2002, 2025-07-02:
>What I meant is that if the frame buffer is used, it's same in the two situations below?

Yes.

closed 2025-07-02 by nopnop2002