nopnop2002/esp-idf-st7789 #54: Fill Screen filling not full screen

AndreyLapshov, 2025-07-19:
Hello. ima using your source . Set it up with menuconfig all is good, fast, but first three functions workink not that i think they should 		FillTest(&dev, CONFIG_WIDTH, CONFIG_HEIGHT); . First fill full screen red, second fill approximately 4/5 . 	
       lcdFillScreen(dev, RED);
	lcdDrawFinish(dev);
	vTaskDelay(50);
	lcdFillScreen(dev, GREEN);
	lcdDrawFinish(dev);
	vTaskDelay(50);
	lcdFillScreen(dev, BLUE);
	lcdDrawFinish(dev);
GREEn and BLUE do this ans i see RED part at the bottom. 
LCD ST7789v3 240x280 
this red bottom part i see looking at all test one by one .

AndreyLapshov, 2025-07-20:
[image]
https://github.com/user-attachments/assets/7b2f3f4a-2c83-499b-a1d2-eb30dc6d0c9a
https://github.com/user-attachments/assets/5e1d09db-0736-436a-a18b-5b6cdf492226
Hello also a bit trace log from devkitc-v4.

AndreyLapshov, 2025-07-20:
Sorry wideo with log wrong i changed vfs.

nopnop2002, 2025-07-20:
You need GRAM Y offset.
Try to set GRAM y offset with menuconfig

AndreyLapshov, 2025-07-20:
Thanx. Solved also I ran lvgl 8.3 over your lib

closed 2025-07-20 by nopnop2002